ARMmbed/mbed-semtech-lora-rf-drivers #28: Duplicate work on the SX1276 driver

helmut64, 2018-07-21:
Dear Team,

I did a lot of work on the SX1276 driver to make it more generic, to support RFM95, MuRata, Heltec and other 1276 based chipsets and to fix issues. What I see, you do similar changes in your repository.

My code works on mbed and as well on Arduino (D21, ESP32 where I developed mbed compatible timers, IO etc, a different topic). I separated the C++ code to ensure it can be used on multiple platforms, e.g. Linux, mbed, Arduino and other environments without changing the core. 

All my changes are available in the SX1276GenericLib on mbed.org. The API is compatible, except for some enhancements to make it easier for protocols to use it.

I use it mainly for our LoRa peer to peer RadioShuttle protocol (see www.radioshuttle.de), however it does not matter if you use LoRaWAN or other protocols, it is basically a link level driver.

I am bringing this up to see if there is a point to join forces for a Generic SX1276 driver, mbed-os is my key development platform, however Arduino is also important, the same for other OS’s like Linux and Windows. My approach covers multiple platforms.

Regards from Hannover Germany.
Helmut Tschemernjak

hasnainvirk, 2018-08-02:
@helmut64 Thanks for reaching out. Your idea looks great and your efforts are commendable. I will discuss it internally and come back to you. Unfortunately, I can't give you a timeline because of less bandwidth available. Keep up the great work and keep in touch.

helmut64, 2018-08-02:
Dear Hasnain Virk, thank you for the feedback, we keep in touch.