ARMmbed/mbed-semtech-lora-rf-drivers #38: Hard Fault while using 5 DIO pins

trowbridgec, 2019-05-20:
We have a custom board that uses a SX1276 controlled by an nRF52840 running the LoRaWAN stack (from Mbed OS). If I configure for using 4 DIO pins (`DIO0`-`DIO3`), the code runs fine; however, if I configure for 5 DIO pins (`DIO0`-`DIO4`), I get an Mbed OS hard fault with an error of "Not allowed in ISR context".

From looking at the implementation of the SX1276 driver, it doesn't look like `DIO4` is used for anything when the modem is in LoRa mode, but I could be missing something. Does the SX1276 driver support 5+ DIO pins in LoRa mode?

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila

trowbridgec, 2019-09-13:
@AnttiKauppila Can you point me to the PR that fixes this issue? Thanks!

AnttiKauppila, 2019-09-16:
https://github.com/ARMmbed/mbed-os/pull/11412

AnttiKauppila, 2019-09-16:
Sorry, forgot to add the link