ARMmbed/mbed-semtech-lora-rf-drivers #39: Read timeout

evandavey, 2019-06-06:
How do I set a read timeout in ms for the receive function?

Does this work on a board where DIO4 and DIO5 are not connected as a skim through the code looked like this were required?

evandavey, 2019-06-07:
I have tested this with a board using a Murata CMWX1ZZABZ module.  Calling radio->receive(3500) the board never times out.

AnttiKauppila, 2019-06-10:
Did you read the information from the header file?
https://github.com/ARMmbed/mbed-semtech-lora-rf-drivers/blob/master/SX1272/SX1272_LoRaRadio.h#L201
That value (3500) is ignored in the call and as receive(void) says you should configure the options using `set_rx_config(parameters)`

evandavey, 2019-06-10:
Yeah, but I guess I don't really understand how to set the timeout.  I assume it is with `symb_timeout` but not sure how this translates into a ms timeout.  I tried a few values in this field but it didn't seem to have any impact.  Which DIO pin is used for the timeout?  I can see the `rx timeout` callback being called in either DIO1 or DIO2 handlers but no obvious place in `receive` where these are set for LoRa (but I don't have a deep understanding of the register calls).

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila