ARMmbed/mbed-semtech-lora-rf-drivers #44: SX1262 0% Downlink Packet Success Rate for LoRaWAN Class C

snelson-senet, 2019-08-13:
Hardware Setup:  NucleoL073RZ + SX1262MB2xAS

Observations:
- Class A: High(>90%) downlink PSR 
- Class C:  low PSR (<10%)  for class A downlinks and 0% for class C downlinks 
- High class A/C PSR using SX1276MB1LAS radio shield
- High class A/C PSR for running latest Semtech LoRaMac-node project

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

closed 2019-09-13 by AnttiKauppila

thendrickson, 2020-03-25:
@snelson-senet I found the bug. Will be issuing a pull request soon.

snelson-senet, 2020-03-25:
@thendrickson That's great news!