ARMmbed/mbed-semtech-lora-rf-drivers #51: Add Semtech drivers configuration as mbed-os component ?

jeromecoutant, 2020-03-09:
Hi

Just wondering why these files are not part of mbed-os ?
As COMPONENT for ex ?

Thx

@0xc0170 
@MarceloSalazar

0xc0170, 2020-03-09:
I don't recall.

cc @AnttiKauppila

AnttiKauppila, 2020-03-09:
It was a decision made while bringing LoRaWAN stack into Mbed OS 5.7. The reason given was to keep Mbed OS as light-weight as possible (if I remember correctly).

jeromecoutant, 2020-03-09:
Goal would be to enable LORA in default configuration for DISCO_L072CZ_LRWAN1

and also for coming STM32WL chips

AnttiKauppila, 2020-03-10:
That is a good reason to bring the drivers into Mbed Os. I need to fix one licence header for SX1262 driver, but after that it needs only a PR to be made for Mbed Os.

jeromecoutant, 2020-03-10:
@LMESTM

AnttiKauppila, 2020-03-10:
Copyright should be fine now

jeromecoutant, 2020-03-25:
@AnttiKauppila any updates ?

kivaisan, 2020-04-01:
PR to include Lora drivers in mbed-os: https://github.com/ARMmbed/mbed-os/pull/12741

kivaisan, 2020-04-17:
PR got merged, so closing this issue. In order to use driver corresponding component must be enabled in mbed_app.json. E.g.:
`"target.components_add": ["SX126X"],`

closed 2020-04-17 by kivaisan