ARMmbed/mbed-semtech-lora-rf-drivers #53: SX1262 times out of rx continuous (class C)

thendrickson, 2020-03-25:
After setting class C, the driver still times out under 1 second on the second RX2 window. I would expect not to see a timeout and stay in continuous receive mode.

thendrickson, 2020-03-25:
@kivaisan Can I send you a patch file of the solution? I wasn't 
able to push the fix to a new branch.

thendrickson, 2020-03-25:
From 748e38d744da9936558ea764e0647e4cd6dddfe8 Mon Sep 17 00:00:00 2001
From: Travis Hendrickson <travis.hendrickson@faithengineeringllc.com>
Date: Wed, 25 Mar 2020 14:33:23 -0600
Subject: [PATCH] BUGFIX: MODEM_LORA rx continuous was timing out

---
 SX126X/SX126X_LoRaRadio.cpp | 23 +++++++++++++----------
 1 file changed, 13 insertions(+), 10 deletions(-)

diff --git a/SX126X/SX126X_LoRaRadio.cpp b/SX126X/SX126X_LoRaRadio.cpp
index 21fa4cf..0ad4e5d 100644
--- a/SX126X/SX126X_LoRaRadio.cpp
+++ b/SX126X/SX126X_LoRaRadio.cpp
@@ -906,7 +906,7 @@ void SX126X_LoRaRadio::set_rx_config(radio_modems_t modem,
         }
 
         case MODEM_LORA: {
-            _rx_timeout_in_symbols = symb_timeout;
+            _rx_timeout_in_symbols = rx_continuous ? 0 : symb_timeout;
             _mod_params.modem_type = MODEM_LORA;
             _mod_params.params.lora.spreading_factor =
                 (lora_spread_factors_t) datarate;
@@ -1002,15 +1002,18 @@ void SX126X_LoRaRadio::send(uint8_t *buffer, uint8_t size)
 
 void SX126X_LoRaRadio::receive(void)
 {
-    if (get_modem() == MODEM_LORA && _reception_mode != RECEPTION_MODE_CONTINUOUS) {
-        // Data-sheet Table 13-11: StopOnPreambParam
-        // We will use radio's internal timer to mark no reception. This behaviour
-        // is different from SX1272/SX1276 where we are relying on radio to stop
-        // at preamble detection.
-        // 0x00 means Timer will be stopped on SyncWord(FSK) or Header (LoRa) detection
-        // 0x01 means Timer is stopped on preamble detection
-        uint8_t stop_at_preamble = 0x01;
-        write_opmode_command(RADIO_SET_STOPRXTIMERONPREAMBLE, &stop_at_preamble, 1);
+    if (get_modem() == MODEM_LORA) {
+        if (_reception_mode != RECEPTION_MODE_CONTINUOUS) {
+            // Data-sheet Table 13-11: StopOnPreambParam
+            // We will use radio's internal timer to mark no reception. This behaviour
+            // is different from SX1272/SX1276 where we are relying on radio to stop
+            // at preamble detection.
+            // 0x00 means Timer will be stopped on SyncWord(FSK) or Header (LoRa) detection
+            // 0x01 means Timer is stopped on preamble detection
+            uint8_t stop_at_preamble = 0x01;
+            write_opmode_command(RADIO_SET_STOPRXTIMERONPREAMBLE, &stop_at_preamble, 1);
+        }
+
         // Data-sheet 13.4.9 SetLoRaSymbNumTimeout
         write_opmode_command(RADIO_SET_LORASYMBTIMEOUT, &_rx_timeout_in_symbols, 1);
     }
-- 
2.23.0

thendrickson, 2020-03-25:
`From 748e38d744da9936558ea764e0647e4cd6dddfe8 Mon Sep 17 00:00:00 2001
From: Travis Hendrickson <travis.hendrickson@faithengineeringllc.com>
Date: Wed, 25 Mar 2020 14:33:23 -0600
Subject: [PATCH] BUGFIX: MODEM_LORA rx continuous was timing out

---
 SX126X/SX126X_LoRaRadio.cpp | 23 +++++++++++++----------
 1 file changed, 13 insertions(+), 10 deletions(-)

diff --git a/SX126X/SX126X_LoRaRadio.cpp b/SX126X/SX126X_LoRaRadio.cpp
index 21fa4cf..0ad4e5d 100644
--- a/SX126X/SX126X_LoRaRadio.cpp
+++ b/SX126X/SX126X_LoRaRadio.cpp
@@ -906,7 +906,7 @@ void SX126X_LoRaRadio::set_rx_config(radio_modems_t modem,
         }
 
         case MODEM_LORA: {
-            _rx_timeout_in_symbols = symb_timeout;
+            _rx_timeout_in_symbols = rx_continuous ? 0 : symb_timeout;
             _mod_params.modem_type = MODEM_LORA;
             _mod_params.params.lora.spreading_factor =
                 (lora_spread_factors_t) datarate;
@@ -1002,15 +1002,18 @@ void SX126X_LoRaRadio::send(uint8_t *buffer, uint8_t size)
 
 void SX126X_LoRaRadio::receive(void)
 {
-    if (get_modem() == MODEM_LORA && _reception_mode != RECEPTION_MODE_CONTINUOUS) {
-        // Data-sheet Table 13-11: StopOnPreambParam
-        // We will use radio's internal timer to mark no reception. This behaviour
-        // is different from SX1272/SX1276 where we are relying on radio to stop
-        // at preamble detection.
-        // 0x00 means Timer will be stopped on SyncWord(FSK) or Header (LoRa) detection
-        // 0x01 means Timer is stopped on preamble detection
-        uint8_t stop_at_preamble = 0x01;
-        write_opmode_command(RADIO_SET_STOPRXTIMERONPREAMBLE, &stop_at_preamble, 1);
+    if (get_modem() == MODEM_LORA) {
+        if (_reception_mode != RECEPTION_MODE_CONTINUOUS) {
+            // Data-sheet Table 13-11: StopOnPreambParam
+            // We will use radio's internal timer to mark no reception. This behaviour
+            // is different from SX1272/SX1276 where we are relying on radio to stop
+            // at preamble detection.
+            // 0x00 means Timer will be stopped on SyncWord(FSK) or Header (LoRa) detection
+            // 0x01 means Timer is stopped on preamble detection
+            uint8_t stop_at_preamble = 0x01;
+            write_opmode_command(RADIO_SET_STOPRXTIMERONPREAMBLE, &stop_at_preamble, 1);
+        }
+
         // Data-sheet 13.4.9 SetLoRaSymbNumTimeout
         write_opmode_command(RADIO_SET_LORASYMBTIMEOUT, &_rx_timeout_in_symbols, 1);
     }
-- 
2.23.0

`

kivaisan, 2020-03-26:
@thendrickson Please fork this repo and then you can make a PR.

thendrickson, 2020-03-26:
PR created

kivaisan, 2020-03-30:
Closing issue as fixed.

closed 2020-03-30 by kivaisan